frankarobotics/franka_ros2 #109: error : MoveGroupInterface::execute() failed or timeout reached

Luoji-0, 2025-02-18:
when I execute ros2 launch franka_fr... , then plan and execute in rviz, it can only plan but execute failed, error:
 MoveGroupInterface::move() failed or timeout reached

Smith-JackSmith, 2025-02-19:
Can you please provide the full command line you are executing.
Also, please confirm the branch of franka_ros2 in use.

whiteNavelOrange, 2025-02-24:
the command：ros2 launch franka_fr3_moveit_config moveit.launch.py robot_ip:=dont-care use_fake_hardware:=true
I ensure that I use the right branch of franka_ros2

whiteNavelOrange, 2025-02-25:
[image],
and I also found that when I execute colcon test,there were some error:
Summary: 12 packages finished [16.3s]
  4 packages had stderr output: franka_example_controllers franka_msgs franka_robot_state_broadcaster joint_trajectory_controller
  4 packages had test failures: franka_example_controllers franka_msgs franka_robot_state_broadcaster joint_trajectory_controller
In the test log:

[image]