frankarobotics/franka_ros2 #110: Is it possible to run MoveIt in Gazebo?

gribes02, 2025-02-26:
Hi, 

I am trying to run both Gazebo and MoveIt at the same time so that I can move the panda arm in Gazebo. I was wondering if that has been implemented? 

For now both MoveIt and Gazebo work separetly. I run Gazebo using this:

```
ros2 launch eut_franka_description visualize_franka_gazebo.launch.py load_gripper:=true franka_hand:='franka_hand'
```

And I can run MoveIt using this with Rviz: 
```
ros2 launch franka_fr3_moveit_config moveit.launch.py robot_ip:=dont-care use_fake_hardware:=true
```

However, running both files at the same time leads to errors as there are several joint publishers. Essentially what I would like to have is a way to move the end-effector in Gazebo. I am aware of the joint position and velocities examples however I would prefer to use the cartesian end effector position instead. 

Thank you!

AndreasKuhner, 2025-03-04:
Hi @gribes02 ,
did you try to comment out the spawning of the joint state publisher in the launch files? (at least in one of them)

Cheers,
Andreas

gribes02, 2025-03-04:
Hi @AndreasKuhner,

Thank you for the reply, yes I have already tried various variations of removing `joint_state_publisher`, `ros2_control_node`, `robot_state_publisher` and `joint_state_broadcaster` in one of the launch files as I saw that it appears in both launch files. However, this leads to rviz complaining that there is no base frame and thus I can't execute movement.

[image]

[image]

I was wondering what would be the best way to move the end effector in gazebo if I can't do it through MoveIt, not using joint positions?

Thanks!

AndreasKuhner, 2025-03-05:
@BarisYazici Do you maybe have an idea?

BarisYazici, 2025-03-05:
maybe you can publish static transform between base and the world 

`ros2 run tf2_ros static_transform_publisher 0 0 0 0 0 0 world base
`

if that works you can add this to your urdf

gribes02, 2025-03-10:
Hi @BarisYazici,

Yes running that command did allow me to move the robot arm in Rviz, and planning does seem to work however when I execute the robot arm does not move. When I run Rviz by itself I do manage to make the arm move. I see that when running the Rviz file I get this error

```
[ros2 run controller_manager spawner fr3_arm_controller-7] "[1741593455.883172525 spawner_fr3_arm_controller 229] Failed loading controller fr3_arm_controller"
```

and when running the rviz file this appears in the gazebo terminal 

```
[ruby $(which ign) gazebo-1] "[1741593491.276219127 controller_manager 712] The 'type' param was not defined for 'fr3_arm_controller'.
```
I see that the fr2 controller is not loading correctly for some reason.

Also I was wondering if I should set  `use_fake_hardware` to true or false because when I set it to true Rviz does appear but when I set it to false it does not. 

I didn't think it would be this tedious to implement Moveit with Gazebo. How does one move the end effector in Gazebo anyways? 

Thank you!

BarisYazici, 2025-03-10:
You don't need MoveIt to move the arm in Gazebo. You can look into our documentation and run the gazebo in joint position(https://frankaemika.github.io/docs/franka_ros2.html#joint-position-control-example-with-gazebo) mode. This will allow a similar control interface to the hardware.

Currently there is no support controlling Gazebo with the MoveIt. But I could give you some hint. Moveit needs joint trajectory controller. And in our JTC config we command with Effort and read the Position and Velocity states. All these states need to be available. You can debug if those state and command interfaces are available with the ros2 control cli: https://control.ros.org/master/doc/ros2_control/ros2controlcli/doc/userdoc.html#command-line-interface
`ros2 control list_hardware_interfaces `

The problem is you need to change the MoveIt config files. They are currently starting the hardware interface of the real robot. So this approach wouldn't work. Maybe it's easier to put the move-group in the gazebo bringups. This way you don't trigger the URDF file which spawns the real hardware interfaces. 

I think with some work you can get it working. But I would suggest you to wait until we support MoveIt with the simulated gazebo robot. I remember MoveIt had some resources they were controlling the IsaacSim simulated franka robot with moveit. Maybe you can look this up: https://moveit.picknik.ai/main/doc/how_to_guides/isaac_panda/isaac_panda_tutorial.html

If you just wanna control the robot in joint position directly sending joint position commands without motion generation and interpolation. You can already do that with our current stack. Try this command this should already start the robot hardware interfaces and the example controller - joint_position_example_controller from the franka_example_controllers package:

`ros2 launch franka_gazebo_bringup gazebo_joint_position_controller_example.launch.py load_gripper:=true franka_hand:='franka_hand' `

closed 2025-03-10 by gribes02

pvmolle, 2025-03-25:
Is Moveit support for the simulated Gazebo robot planned? What would be the best approach to control the simulated robot using cartesian end effector goals in the meantime?