frankarobotics/franka_ros2 #115: Why is it that when starting this file joint_impedance_with_ik_example_controller.launch.py with virtual hardware, the screen says that it cannot connect to the service

vegetabledogzy, 2025-03-06:
Why is it that when starting this file joint_impedance_with_ik_example_controller.launch.py with virtual hardware, the screen says that it cannot connect to the service
`ros2 launch franka_bringup joint_velocity_example_controller.launch.py arm_id:=fr3 robot_ip:=don use_fake_hardware:=true use_rviz:=true` 

can't connect
`[ros2_control_node-3] [INFO] [1741268200.061171881] [joint_velocity_example_controller]: Extracted Robot Name: fr3
[INFO] [spawner-4]: process has finished cleanly [pid 212358]
[spawner-7] [WARN] [1741268210.059234571] [spawner_joint_velocity_example_controller]: Failed getting a result from calling /controller_manager/configure_controller in 10.0. (Attempt 1 of 3.)
`

vegetabledogzy, 2025-03-06:
When I run `ros2 launch franka_bringup franka.launch.py arm_id:=fr3 robot_ip:=don use_fake_hardware:=true use_rviz:=true
` everything works fine. Then when I use `ros2 control load_controller --set-state active joint_velocity_example_controller
`, there will be a loss of connection.`zy@zy-System-Product-Name:~/dev1/franka_ros2_ws$ ros2 control load_controller --set-state active joint_velocity_example_controller
[WARN] [1741268700.064793874] [_ros2cli_215856]: Failed getting a result from calling /controller_manager/configure_controller in 10.0. (Attempt 1 of 3.)
`

vegetabledogzy, 2025-03-06:
Could not successfully call service /controller_manager/configure_controller after 3 attempts.

vegetabledogzy, 2025-03-07:
I solved the problem.To register the controller, configure the controller's yaml file in the config of the corresponding feature package.

AndreasKuhner, 2025-03-10:
Hi @vegetabledogzy ,
if you just use fake hardware, there is not 'much' running which serves as a backend for running a controller. For that you would need to run a simulation.

Cheers,
Andreas