frankarobotics/franka_ros2 #119: MultiFrankaHardwareInterface Does Not Exist

jcarpinelli-bdai, 2025-03-10:
After loading using the new `multi_arm` feature of  `franka_description`, the following plugin fails to load: `franka_hardware/MultiFrankaHardwareInterface`. After doing some searching, I cannot find any instance of `MultiFrankaHardwareInterface` in `franka_hardware` on `humble` or `develop`. Where can I find the hardware interface required for running multiple arms at once?

BarisYazici, 2025-03-10:
Hmm you are right. That's a mistake from our side. Currently multi_arm is not supported with ros2 control.

jcarpinelli-bdai, 2025-03-10:
Thanks for the quick reply. Is `ros2_control` support planned?

AndreasKuhner, 2025-03-10:
Yes, it is planned 😄

kmohyeldin, 2025-03-11:
@jcarpinelli-bdai what control interfaces are you using?

B-ijan, 2025-03-11:
Hi,
great to hear you're working on that! Do you have a rough timeline for when multi-arm support will be available?
In the meantime, do you have any recommendations for a workaround?
Looking forward to your insights!

jcarpinelli-bdai, 2025-03-11:
> @jcarpinelli-bdai what control interfaces are you using?

@kmohyeldin we are using effort command interfaces for Franka FR3 arms. Unfortunately the sensitivity documented by #55 prevents us from using position control.

jcarpinelli-bdai, 2025-03-18:
> For sensitivity issue, it will take a bit more time as it requires some system image update.

This sounds interesting, and would be high impact! Could you clarify what behavior may be changing to address the motion sensitivity?

Elise-J, 2025-05-02:
Hello,
I would also be interessed in being able to use multiple franka arm.
Is there any update on this? Or a quick workaround? 

Thank you!

AndreasKuhner, 2025-05-26:
Hi @Elise-J ,
we are working on it - currently in the merging/cleaning up phase. Stay tuned 🥳 

Cheers,
Andreas

jcarpinelli-bdai, 2025-06-11:
Great news, thanks for the update! We've been using a similar workflow for two-arm setups. The `arm_prefix` argument has been incredibly useful, I'm glad y'all added that. :)