frankarobotics/franka_ros2 #120: Best practice for changing the end-effector?

pauljansonnie, 2025-03-12:
Hello,

I attached a camera mount between the flange and the hand of the robot (FR3). As a consequence, the hand is offset by a few millimeters, and the same goes for the default TCP. I would like to update the TCP such that this offset is reflected when controlling the end-effector AND when reading the state of the robot (more precisely the pose of the end-effector). I am using a controller based on https://github.com/frankaemika/franka_ros2/blob/humble/franka_example_controllers/src/cartesian_pose_example_controller.cpp with very little modifications.

I would like to know what is best practice to do so. For now, what I did is
1) updating the URDF by updating joint8 in https://github.com/frankaemika/franka_description/blob/main/robots/fr3/kinematics.yaml#L69 (increasing z-value by a few mm)
2) using the service `/service_server/set_tcp_frame` to reflect the very same offset.

Am I missing something (maybe with `/service_server/set_stiffness_frame`)? Is there a better way to do this? A tiny issue with this approach is that I must manually make sure that both z-values match.

Any help is welcome. It can also become useful if I decide to use another gripper in the future. Thank you in advance.

AndreasKuhner, 2025-03-12:
Hi @pauljansonnie ,
I guess you already said a few of your options:
- Either adapt the URDF and set the tcp frame. Therefore, you should be able to control the robot via Cartesian in the TCP frame
- Or handle the transformations yourself and still command in the flange frame
- Or configure your new gripper setup in Desk via the end-effector configuration. Then again, you can control the robot your Cartesian motion. Unfortunately, this change is not propagated to ROS 2, hence you would still to manually need to change the URDF

And I guess there are more options 🤔 At least I am not aware of best practices - whatever works :)

Cheers,
Andreas

pauljansonnie, 2025-03-17:
Thank you Andreas!

closed 2025-03-17 by pauljansonnie