frankarobotics/franka_ros2 #128: Use robot_description topic instead of parameter for Franka controllers

jellehierck, 2025-04-07:
The robot description for controllers and the ControllerManager should be obtained from the `/robot_description` topic instead of being passed as a parameter.

This issue continues on https://github.com/frankaemika/franka_ros2/pull/112:

> Since https://github.com/ros-controls/ros2_control/pull/940, passing a URDF file directly to the controller_manager is deprecated. Instead, the robot_state_publisher makes the robot description available on topic /robot_description and the controller manager should subscribe to that topic.
> 
> Every launch file now emits a warning:
> 
> [ros2_control_node-3] [WARN] [1741096056.232298456] [controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' 

It turns out we cannot simply remove the `robot_description` parameter from the ControllerManager since it is required by loaded Franka controllers, and passed to those controllers by setting the parameter on the ControllerManager in `franka.launch.py` ([here](https://github.com/frankaemika/franka_ros2/blob/humble/franka_bringup/launch/franka.launch.py#L80)).

In the discussion on https://github.com/frankaemika/franka_ros2/pull/112, @AndreasKuhner suggested to obtain the robot description inside the Franka controllers from a subscription to topic `/robot_description`, similar to the ControllerManager itself.

jellehierck, 2025-04-07:
A similar question was asked in the `ros2_control` issues: https://github.com/ros-controls/ros2_control/issues/1450. Apparently, this issue is already fixed from Jazzy onwards, since the robot URDF is available as an attribute to all controllers from the ControllerManager (https://github.com/ros-controls/ros2_control/pull/1088). 

This change will _not_ be backported to Humble according to https://github.com/ros-controls/ros2_control/issues/1450#issuecomment-2006707846. So if Franka ROS 2 is moving to Jazzy in the near future, this issue could be fixed automatically.

AndreasKuhner, 2025-04-07:
Thanks @jellehierck for the research. 

Then I guess we stay for humble with the deprecation warning and adapt with Jazzy.