frankarobotics/franka_ros2 #137: Failed to launch 3rd robot into ros2 network with node launch collapse

dp-tumLi, 2025-05-27:
We have a multi-robot ros2 system, we face the problem that the 3rd robot (dependent on the order we launch) gets difficult to joint in the network with collapsed node launching as shown the error below, which indicates _one controller was already loaded_. As each robot will bring up three nodes, it can be possible that arbitrary of the nodes will fail, following just show one of the cases. And, if **we launch more robots at the same time**, we also meet this error in each robot. 

Is this something issue of the way we define our controller name space? we are certain we make different name space for different robot, and they are clarified differently as well. Or would it be related to network quality, dds setting, etc??

`[ros2_control_node-3] [INFO] [1748346856.116982031] [right_arm.controller_manager]: Loading controller 'joint_state_broadcaster3'
[ros2_control_node-3] [INFO] [1748346860.400823599] [right_arm.controller_manager]: Loading controller 'franka_robot_state_broadcaster3'
[spawner-4] [INFO] [1748346860.402406520] [right_arm.spawner_joint_state_broadcaster3]: Loaded joint_state_broadcaster3
[spawner-5] [WARN] [1748346866.116751171] [right_arm.spawner_franka_robot_state_broadcaster3]: Failed getting a result from calling /right_arm/controller_manager/load_controller in 10.0. (Attempt 1 of 3.)
[spawner-6] [WARN] [1748346866.118079234] [right_arm.spawner_joint_velocity_rcm_controller3]: Failed getting a result from calling /right_arm/controller_manager/load_controller in 10.0. (Attempt 1 of 3.)
[ros2_control_node-3] [INFO] [1748346868.111867448] [right_arm.controller_manager]: Loading controller 'joint_velocity_rcm_controller3'
[spawner-4] [WARN] [1748346870.404778133] [right_arm.spawner_joint_state_broadcaster3]: Failed getting a result from calling /right_arm/controller_manager/configure_controller in 10.0. (Attempt 1 of 3.)
[spawner-5] [WARN] [1748346876.117622538] [right_arm.spawner_franka_robot_state_broadcaster3]: Failed getting a result from calling /right_arm/controller_manager/load_controller in 10.0. (Attempt 2 of 3.)
[spawner-6] [WARN] [1748346876.119016464] [right_arm.spawner_joint_velocity_rcm_controller3]: Failed getting a result from calling /right_arm/controller_manager/load_controller in 10.0. (Attempt 2 of 3.)
[ros2_control_node-3] [INFO] [1748346876.823017459] [right_arm.controller_manager]: Loading controller 'franka_robot_state_broadcaster3'
[ros2_control_node-3] [ERROR] [1748346876.823111398] [right_arm.controller_manager]: A controller named 'franka_robot_state_broadcaster3' was already loaded inside the controller manager
[ros2_control_node-3] [INFO] [1748346876.823375753] [right_arm.controller_manager]: Loading controller 'joint_velocity_rcm_controller3'
[ros2_control_node-3] [ERROR] [1748346876.823440707] [right_arm.controller_manager]: A controller named 'joint_velocity_rcm_controller3' was already loaded inside the controller manager
[ros2_control_node-3] [INFO] [1748346876.823596387] [right_arm.controller_manager]: Loading controller 'franka_robot_state_broadcaster3'
[ros2_control_node-3] [ERROR] [1748346876.823628864] [right_arm.controller_manager]: A controller named 'franka_robot_state_broadcaster3' was already loaded inside the controller manager
[ros2_control_node-3] [INFO] [1748346876.823785312] [right_arm.controller_manager]: Loading controller 'joint_velocity_rcm_controller3'
[ros2_control_node-3] [ERROR] [1748346876.823905512] [right_arm.controller_manager]: A controller named 'joint_velocity_rcm_controller3' was already loaded inside the controller manager
[ros2_control_node-3] [INFO] [1748346876.824084589] [right_arm.controller_manager]: Configuring controller 'joint_state_broadcaster3'
[ros2_control_node-3] [INFO] [1748346876.824183277] [right_arm.joint_state_broadcaster3]: 'joints' or 'interfaces' parameter is empty. All available state interfaces will be published
[spawner-5] [FATAL] [1748346876.824674413] [right_arm.spawner_franka_robot_state_broadcaster3]: Failed loading controller franka_robot_state_broadcaster3
[spawner-6] [FATAL] [1748346876.824812004] [right_arm.spawner_joint_velocity_rcm_controller3]: Failed loading controller joint_velocity_rcm_controller3
[ros2_control_node-3] [INFO] [1748346876.842876812] [right_arm.controller_manager]: Configuring controller 'joint_state_broadcaster3'
[ros2_control_node-3] [INFO] [1748346876.843035496] [right_arm.joint_state_broadcaster3]: 'joints' or 'interfaces' parameter is empty. All available state interfaces will be published
[ERROR] [spawner-5]: process has died [pid 1040570, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner franka_robot_state_broadcaster3 --ros-args -r __ns:=/right_arm'].
[ERROR] [spawner-6]: process has died [pid 1040572, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner joint_velocity_rcm_controller3 --ros-args -r __ns:=/right_arm'].
[spawner-4] [INFO] [1748346877.330314660] [right_arm.spawner_joint_state_broadcaster3]: Configured and activated joint_state_broadcaster3
[INFO] [spawner-4]: process has finished cleanly [pid 1040568]`

AndreasKuhner, 2025-05-28:
Hi @dp-tumLi ,
in your log is not much to see - but it at least says that you have some duplicates. E.g.` A controller named 'joint_velocity_rcm_controller3' was already loaded inside the controller manager`.

Maybe you need to check that?

Cheers,
Andreas

AndreasKuhner, 2025-05-28:
I think if you use different namespaces and it still complains about already loaded controllers, something with your namespaces might be off. Maybe you can somehow print all the names you start and check if those are the expected ones.