frankarobotics/franka_ros2 #145: Problem when launching franka_bringup

rikhsitlladeveloper, 2025-06-17:
Hello I am using Ros2 Humble, with libfranka 0.15.0 , Robot Control is 5.8.0 , franka_ros2 v1.0.0. 

When I launch ros2 launch franka_bringup cartesian_pose_example_controller.launch.py 
Or other launch files, there is following error.

 ros2 launch franka_bringup cartesian_pose_example_controller.launch.py 
[INFO] [launch]: All log files can be found below /home/user/.ros/log/2025-06-17-05-07-51-335905-franka-ai-companion-640451
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [joint_state_publisher-1]: process started with pid [640465]
[INFO] [robot_state_publisher-2]: process started with pid [640467]
[INFO] [ros2_control_node-3]: process started with pid [640469]
[INFO] [spawner-4]: process started with pid [640471]
[INFO] [spawner-5]: process started with pid [640473]
[INFO] [spawner-6]: process started with pid [640475]
[ros2_control_node-3] [WARN] [1750136871.796107330] [controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ros2_control_node-3] text not specified in the prefix tag
[ros2_control_node-3] text not specified in the arm_prefix tag
[ros2_control_node-3] [INFO] [1750136871.796825587] [resource_manager]: Loading hardware 'FrankaHardwareInterface' 
[ros2_control_node-3] [INFO] [1750136871.813566937] [resource_manager]: Initialize hardware 'FrankaHardwareInterface' 
[ros2_control_node-3] [INFO] [1750136871.813726525] [FrankaHardwareInterface]: Connecting to robot at "192.168.19.2" ...
[robot_state_publisher-2] [INFO] [1750136871.829915574] [robot_state_publisher]: got segment base
[ros2_control_node-3] [INFO] [1750136871.830008408] [FrankaHardwareInterface]: Successfully connected to robot
[robot_state_publisher-2] [INFO] [1750136871.830191900] [robot_state_publisher]: got segment fr3_link0
[robot_state_publisher-2] [INFO] [1750136871.830221885] [robot_state_publisher]: got segment fr3_link1
[robot_state_publisher-2] [INFO] [1750136871.830253406] [robot_state_publisher]: got segment fr3_link2
[robot_state_publisher-2] [INFO] [1750136871.830261598] [robot_state_publisher]: got segment fr3_link3
[robot_state_publisher-2] [INFO] [1750136871.830268574] [robot_state_publisher]: got segment fr3_link4
[robot_state_publisher-2] [INFO] [1750136871.830275486] [robot_state_publisher]: got segment fr3_link5
[robot_state_publisher-2] [INFO] [1750136871.830282110] [robot_state_publisher]: got segment fr3_link6
[robot_state_publisher-2] [INFO] [1750136871.830288671] [robot_state_publisher]: got segment fr3_link7
[robot_state_publisher-2] [INFO] [1750136871.830294943] [robot_state_publisher]: got segment fr3_link8
[ros2_control_node-3] [INFO] [1750136871.846378550] [service_server]: Service started
[ros2_control_node-3] [INFO] [1750136871.981600870] [action_server]: Action server started
[ros2_control_node-3] [INFO] [1750136871.981785898] [resource_manager]: Successful initialization of hardware 'FrankaHardwareInterface'
[ros2_control_node-3] [INFO] [1750136871.982093649] [resource_manager]: 'configure' hardware 'FrankaHardwareInterface' 
[ros2_control_node-3] [INFO] [1750136871.982118898] [resource_manager]: Successful 'configure' of hardware 'FrankaHardwareInterface'
[ros2_control_node-3] [INFO] [1750136871.982144243] [resource_manager]: 'activate' hardware 'FrankaHardwareInterface' 
[ros2_control_node-3] [INFO] [1750136871.982709024] [FrankaHardwareInterface]: Started
[ros2_control_node-3] [INFO] [1750136871.982738720] [resource_manager]: Successful 'activate' of hardware 'FrankaHardwareInterface'
[ros2_control_node-3] [INFO] [1750136872.166073394] [controller_manager]: update rate is 1000 Hz
[ros2_control_node-3] [INFO] [1750136872.166264503] [controller_manager]: Spawning controller_manager RT thread with scheduler priority: 50
[ros2_control_node-3] [INFO] [1750136872.179320711] [controller_manager]: Successful set up FIFO RT scheduling policy with priority 50.
[ros2_control_node-3] [INFO] [1750136872.368253276] [controller_manager]: Loading controller 'franka_robot_state_broadcaster'
[joint_state_publisher-1] [INFO] [1750136872.374217991] [joint_state_publisher]: Waiting for robot_description to be published on the robot_description topic...
[ros2_control_node-3] [INFO] [1750136872.397989969] [controller_manager]: Loading controller 'cartesian_pose_example_controller'
[ros2_control_node-3] [INFO] [1750136872.421432148] [controller_manager]: Loading controller 'joint_state_broadcaster'
[spawner-5] [INFO] [1750136872.424871652] [spawner_franka_robot_state_broadcaster]: Loaded franka_robot_state_broadcaster
[ros2_control_node-3] [INFO] [1750136872.442392348] [controller_manager]: Configuring controller 'franka_robot_state_broadcaster'
[spawner-6] [INFO] [1750136872.449321470] [spawner_cartesian_pose_example_controller]: Loaded cartesian_pose_example_controller
[spawner-4] [INFO] [1750136872.472538363] [spawner_joint_state_broadcaster]: Loaded joint_state_broadcaster
[ros2_control_node-3] [INFO] [1750136872.515847917] [controller_manager]: Configuring controller 'cartesian_pose_example_controller'
[ros2_control_node-3] [INFO] [1750136872.521131688] [cartesian_pose_example_controller]: Default collision behavior set.
[ros2_control_node-3] [INFO] [1750136872.533258595] [cartesian_pose_example_controller]: Extracted Robot Name: fr3
[ros2_control_node-3] [INFO] [1750136872.547329771] [controller_manager]: Configuring controller 'joint_state_broadcaster'
[ros2_control_node-3] [INFO] [1750136872.547484334] [joint_state_broadcaster]: 'joints' or 'interfaces' parameter is empty. All available state interfaces will be published
[spawner-5] [INFO] [1750136872.567388702] [spawner_franka_robot_state_broadcaster]: Configured and activated franka_robot_state_broadcaster
[ros2_control_node-3] terminate called after throwing an instance of 'franka::ControlException'
[ros2_control_node-3]   what():  libfranka: Move command rejected: cannot start at singular pose!
[ros2_control_node-3] Stack trace (most recent call last) in thread 640531:
[ros2_control_node-3] #20   Object "/usr/lib/aarch64-linux-gnu/ld-linux-aarch64.so.1", at 0xffffffffffffffff, in 
[ros2_control_node-3] #19   Object "/usr/lib/aarch64-linux-gnu/libc.so.6", at 0xffffa7ee5edb, in 
[ros2_control_node-3] #18   Object "/usr/lib/aarch64-linux-gnu/libc.so.6", at 0xffffa7e7d5b7, in 
[ros2_control_node-3] #17   Object "/usr/lib/aarch64-linux-gnu/libstdc++.so.6.0.30", at 0xffffa80b31fb, in 
[ros2_control_node-3] #16   Object "/opt/ros/humble/lib/controller_manager/ros2_control_node", at 0xaaaac54d5b5b, in 
[ros2_control_node-3] #15   Object "/opt/ros/humble/lib/libcontroller_manager.so", at 0xffffa8507de7, in controller_manager::ControllerManager::update(rclcpp::Time const&, rclcpp::Duration const&)
[ros2_control_node-3] #14   Object "/opt/ros/humble/lib/libcontroller_manager.so", at 0xffffa84fa877, in controller_manager::ControllerManager::manage_switch()
[ros2_control_node-3] #13   Object "/opt/ros/humble/lib/libhardware_interface.so", at 0xffffa7cac40b, in hardware_interface::ResourceManager::perform_command_mode_switch(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&)
[ros2_control_node-3] #12   Object "/opt/ros/humble/lib/libhardware_interface.so", at 0xffffa7cac243, in 
[ros2_control_node-3] #11   Object "/ros2_ws/build/franka_hardware/libfranka_hardware.so", at 0xffffa077448b, in franka_hardware::FrankaHardwareInterface::perform_command_mode_switch(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&)
[ros2_control_node-3] #10   Object "/ros2_ws/build/franka_hardware/libfranka_hardware.so", at 0xffffa078316b, in franka_hardware::Robot::initializeCartesianPoseInterface()
[... 4 lines cut ...]
[... cut ...]
[ERROR] [spawner-4]: process[spawner-4] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[INFO] [spawner-6]: sending signal 'SIGTERM' to process[spawner-6]
[INFO] [spawner-4]: sending signal 'SIGTERM' to process[spawner-4]
[ERROR] [spawner-6]: process has died [pid 640475, exit code -15, cmd '/opt/ros/humble/lib/controller_manager/spawner cartesian_pose_example_controller --ros-args'].
[ERROR] [spawner-4]: process has died [pid 640471, exit code -15, cmd '/opt/ros/humble/lib/controller_manager/spawner joint_state_broadcaster --ros-args'].

Pls any recoomendation will help me.

rikhsitlladeveloper, 2025-06-27:
@AndreasKuhner  After using new releases it is working, But why less launch files in franka_bringup ?
I need Cartesian Impedance controller with Moveit2.
Thanks

AndreasKuhner, 2025-07-03:
We unified the launch file to work with almost all standard examples. E.g. you can now do a 
`ros2 launch franka_bringup example.launch.py controller_name:=joint_position_example_controller`

... just add whatever example you wanna run.

Cheers,
Andreas

AndreasKuhner, 2025-07-03:
Additional note: The parameters can now be put into the franka.config.yaml - for convenience.

closed 2025-08-01 by ignacio-pm