frankarobotics/franka_ros2 #146: Problem Launching Franka_bringup

bk-poudel, 2025-06-17:
When, I try to run ros2 launch franka_bringup example.launch.py controller_name:=JointPositionExampleController

I get error, user@bibek-Lenovo-Slim-7-14IRP8:/ros2_ws$ ros2 launch franka_bringup example.launch.py controller_name:=JointPositionExampleController
[INFO] [launch]: All log files can be found below /home/user/.ros/log/2025-06-17-07-32-12-034125-bibek-Lenovo-Slim-7-14IRP8-22327
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [22328]
[INFO] [ros2_control_node-2]: process started with pid [22330]
[INFO] [joint_state_publisher-3]: process started with pid [22332]
[INFO] [spawner-4]: process started with pid [22334]
[INFO] [spawner-5]: process started with pid [22336]
[INFO] [spawner-6]: process started with pid [22338]
[ros2_control_node-2] [WARN] [1750145532.214497860] [NS_1.controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ros2_control_node-2] text not specified in the prefix tag
[ros2_control_node-2] text not specified in the arm_prefix tag
[ros2_control_node-2] [INFO] [1750145532.214901887] [resource_manager]: Loading hardware 'FrankaHardwareInterface' 
[ros2_control_node-2] [INFO] [1750145532.226858825] [resource_manager]: Initialize hardware 'FrankaHardwareInterface' 
[ros2_control_node-2] Parameter 'version' is not set. Please update your URDF (aka franka_description).
**_### [ros2_control_node-2] [FATAL] [1750145532.226993576] [FrankaHardwareInterface]: Parameter '' is not set. Please update your URDF (aka franka_description)._**
[ros2_control_node-2] [INFO] [1750145532.227013606] [resource_manager]: Failed to initialize hardware 'FrankaHardwareInterface'
[ros2_control_node-2] [WARN] [1750145532.227020894] [resource_manager]: System hardware component 'FrankaHardwareInterface' from plugin 'franka_hardware/FrankaHardwareInterface' failed to initialize.
[ros2_control_node-2] terminate called after throwing an instance of 'std::runtime_error'
[ros2_control_node-2]   what():  Wrong state or command interface configuration.
[ros2_control_node-2] missing state interfaces:
[ros2_control_node-2] ' fr3_joint1/position '   ' fr3_joint1/velocity ' ' fr3_joint1/effort '       ' fr3_joint2/position ' ' fr3_joint2/velocity ' ' fr3_joint2/effort '       ' fr3_joint3/position ' ' fr3_joint3/velocity ' ' fr3_joint3/effort '   ' fr3_joint4/position '     ' fr3_joint4/velocity ' ' fr3_joint4/effort '   ' fr3_joint5/position '     ' fr3_joint5/velocity ' ' fr3_joint5/effort '   ' fr3_joint6/position '     ' fr3_joint6/velocity ' ' fr3_joint6/effort '   ' fr3_joint7/position ' ' fr3_joint7/velocity '     ' fr3_joint7/effort '
[ros2_control_node-2] missing command interfaces:
[ros2_control_node-2] ' fr3_joint1/position '   ' fr3_joint1/velocity ' ' fr3_joint1/effort '       ' fr3_joint2/position ' ' fr3_joint2/velocity ' ' fr3_joint2/effort '       ' fr3_joint3/position ' ' fr3_joint3/velocity ' ' fr3_joint3/effort '   ' fr3_joint4/position '     ' fr3_joint4/velocity ' ' fr3_joint4/effort '   ' fr3_joint5/position '     ' fr3_joint5/velocity ' ' fr3_joint5/effort '   ' fr3_joint6/position '     ' fr3_joint6/velocity ' ' fr3_joint6/effort '   ' fr3_joint7/position ' ' fr3_joint7/velocity '     ' fr3_joint7/effort '
[ros2_control_node-2] Stack trace (most recent call last):
[ros2_control_node-2] #15   Object "", at 0xffffffffffffffff, in 
[ros2_control_node-2] #14   Object "/opt/ros/humble/lib/controller_manager/ros2_control_node", at 0x61314a039ee4, in 
[ros2_control_node-2] #13   Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x702a25c32e3f, in __libc_start_main
[ros2_control_node-2] #12   Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x702a25c32d8f, in 
[ros2_control_node-2] #11   Object "/opt/ros/humble/lib/controller_manager/ros2_control_node", at 0x61314a0393e3, in 
[ros2_control_node-2] #10   Object "/opt/ros/humble/lib/libcontroller_manager.so", at 0x702a2634774a, in controller_manager::ControllerManager::ControllerManager(std::shared_ptr<rclcpp::Executor>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::NodeOptions const&)
[ros2_control_node-2] #9    Object "/opt/ros/humble/lib/libcontroller_manager.so", at 0x702a263426df, in controller_manager::ControllerManager::init_resource_manager(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[ros2_control_node-2] #8    Object "/opt/ros/humble/lib/libhardware_interface.so", at 0x702a25aff334, in hardware_interface::ResourceManager::load_urdf(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, bool)
[ros2_control_node-2] #7    Object "/opt/ros/humble/lib/libhardware_interface.so", at 0x702a25adb723, in 
[ros2_control_node-2] #6    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x702a25f004d7, in __cxa_throw
[ros2_control_node-2] #5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x702a25f00276, in std::terminate()
[ros2_control_node-2] #4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x702a25f0020b, in 
[ros2_control_node-2] #3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x702a25ef4b9d, in 
[ros2_control_node-2] #2    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x702a25c317f2, in abort
[ros2_control_node-2] #1    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x702a25c4b475, in raise
[ros2_control_node-2] #0    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x702a25c9f9fc, in pthread_kill
[ros2_control_node-2] Aborted (Signal sent by tkill() 22330 1000)
[robot_state_publisher-1] [INFO] [1750145532.239952783] [NS_1.robot_state_publisher]: got segment base
[robot_state_publisher-1] [INFO] [1750145532.240095184] [NS_1.robot_state_publisher]: got segment fr3_link0
[robot_state_publisher-1] [INFO] [1750145532.240110041] [NS_1.robot_state_publisher]: got segment fr3_link1
[robot_state_publisher-1] [INFO] [1750145532.240119925] [NS_1.robot_state_publisher]: got segment fr3_link2
[robot_state_publisher-1] [INFO] [1750145532.240128784] [NS_1.robot_state_publisher]: got segment fr3_link3
[robot_state_publisher-1] [INFO] [1750145532.240137215] [NS_1.robot_state_publisher]: got segment fr3_link4
[robot_state_publisher-1] [INFO] [1750145532.240145558] [NS_1.robot_state_publisher]: got segment fr3_link5
[robot_state_publisher-1] [INFO] [1750145532.240153757] [NS_1.robot_state_publisher]: got segment fr3_link6
[robot_state_publisher-1] [INFO] [1750145532.240161545] [NS_1.robot_state_publisher]: got segment fr3_link7
[robot_state_publisher-1] [INFO] [1750145532.240169493] [NS_1.robot_state_publisher]: got segment fr3_link8
[spawner-4] [INFO] [1750145532.505637628] [NS_1.spawner_joint_state_broadcaster]: waiting for service /NS_1/controller_manager/list_controllers to become available...
[spawner-5] [INFO] [1750145532.555416961] [NS_1.spawner_franka_robot_state_broadcaster]: waiting for service /NS_1/controller_manager/list_controllers to become available...
[spawner-6] [INFO] [1750145532.555417134] [NS_1.spawner_JointPositionExampleController]: waiting for service /NS_1/controller_manager/list_controllers to become available...
[ERROR] [ros2_control_node-2]: process has died [pid 22330, exit code -6, cmd '/opt/ros/humble/lib/controller_manager/ros2_control_node --ros-args -r __ns:=/NS_1 --params-file /ros2_ws/install/franka_bringup/share/franka_bringup/config/controllers.yaml --params-file /tmp/launch_params_wt6504wh --params-file /tmp/launch_params_j9axufr4'].
[INFO] [launch]: process[ros2_control_node-2] was required: shutting down launched system
[INFO] [spawner-6]: sending signal 'SIGINT' to process[spawner-6]
[INFO] [spawner-5]: sending signal 'SIGINT' to process[spawner-5]
[INFO] [spawner-4]: sending signal 'SIGINT' to process[spawner-4]
[INFO] [joint_state_publisher-3]: sending signal 'SIGINT' to process[joint_state_publisher-3]
[INFO] [robot_state_publisher-1]: sending signal 'SIGINT' to process[robot_state_publisher-1]
[joint_state_publisher-3] Traceback (most recent call last):
[joint_state_publisher-3]   File "/opt/ros/humble/lib/joint_state_publisher/joint_state_publisher", line 33, in <module>
[... cut ...]
[ERROR] [joint_state_publisher-3]: process has died [pid 22332, exit code -2, cmd '/opt/ros/humble/lib/joint_state_publisher/joint_state_publisher --ros-args -r __node:=joint_state_publisher -r __ns:=/NS_1 --params-file /tmp/launch_params_19om_di_'].
[INFO] [robot_state_publisher-1]: process has finished cleanly [pid 22328]
[INFO] [spawner-6]: process has finished cleanly [pid 22338]
[INFO] [spawner-4]: process has finished cleanly [pid 22334]
[INFO] [spawner-5]: process has finished cleanly [pid 22336]
Any help is appreciated

gomezgu, 2025-06-20:
Hi @bk-poudel,

The problem with franka_description should be solved as we released a new version today.

VaRuN-2509, 2025-06-25:
The change I did was
-> cd franka_description
-> git fetch
->git checkout 0.5.1

closed 2025-08-01 by ignacio-pm