frankarobotics/franka_ros2 #154: The Rviz+Moveit crashes after sometime

VaRuN-2509, 2025-06-26:
I ran the command ros2 launch franka_fr3_moveit_config moveit.launch.py robot_ip:=<fci-ip> with te correct robot ip,it works fine but after 2 minutes it crashes suddenly and gets erminated eventhough i didnt do anything(didnt plan any trajectory using moveit).

[rviz2-1] [WARN] [1750927470.017073329] [moveit_robot_model.robot_model]: Skipping virtual joint 'virtual_joint' because its child frame 'fr3_link0' does not match the URDF frame 'base'
[rviz2-1] [INFO] [1750927470.017075465] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ros2_control_node-4] [INFO] [1750927470.047017562] [controller_manager]: Loading controller 'franka_robot_state_broadcaster'
[spawner-6] [INFO] [1750927470.053305698] [spawner_franka_robot_state_broadcaster]: Loaded franka_robot_state_broadcaster
[ros2_control_node-4] [INFO] [1750927470.053953615] [controller_manager]: Configuring controller 'franka_robot_state_broadcaster'
[spawner-6] [INFO] [1750927470.074079229] [spawner_franka_robot_state_broadcaster]: Configured and activated franka_robot_state_broadcaster
[INFO] [ros2-9]: process has finished cleanly [pid 4739]
[INFO] [spawner-6]: process has finished cleanly [pid 4733]
[rviz2-1] [ERROR] [1750927473.290714837] [moveit_ros_visualization.motion_planning_frame]: Action server: /recognize_objects not available
[rviz2-1] [INFO] [1750927473.374457979] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00116921 seconds
[rviz2-1] [INFO] [1750927473.374484220] [moveit_robot_model.robot_model]: Loading robot model 'fr3'...
[rviz2-1] [WARN] [1750927473.374487666] [moveit_robot_model.robot_model]: Skipping virtual joint 'virtual_joint' because its child frame 'fr3_link0' does not match the URDF frame 'base'
[rviz2-1] [INFO] [1750927473.374491390] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[rviz2-1] [INFO] [1750927473.406571388] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[rviz2-1] [INFO] [1750927473.406969773] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/monitored_planning_scene'
[rviz2-1] [INFO] [1750927476.340077035] [moveit_ros_visualization.motion_planning_frame]: MoveGroup namespace changed: / -> . Reloading params.
[rviz2-1] [INFO] [1750927476.434837818] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[rviz2-1] [INFO] [1750927476.435631501] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/monitored_planning_scene'
[rviz2-1] [INFO] [1750927476.445869432] [interactive_marker_display_95786244975872]: Connected on namespace: /rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic
[rviz2-1] [INFO] [1750927476.451533447] [moveit_ros_visualization.motion_planning_frame]: group fr3_manipulator
[rviz2-1] [INFO] [1750927476.451545925] [moveit_ros_visualization.motion_planning_frame]: Constructing new MoveGroup connection for group 'fr3_manipulator' in namespace ''
[rviz2-1] [INFO] [1750927476.457658455] [move_group_interface]: Ready to take commands for planning group fr3_manipulator.
[rviz2-1] [INFO] [1750927476.467173383] [interactive_marker_display_95786244975872]: Sending request for interactive markers
[rviz2-1] [INFO] [1750927476.498604897] [interactive_marker_display_95786244975872]: Service response received for initialization
[ros2_control_node-4] terminate called after throwing an instance of 'franka::ControlException'
[ros2_control_node-4]   what():  libfranka: Move command aborted: motion aborted by reflex! ["communication_constraints_violation"]
[ros2_control_node-4] Stack trace (most recent call last) in thread 4911:
[ros2_control_node-4] #19   Object "", at 0xffffffffffffffff, in 
[ros2_control_node-4] #18   Source "../sysdeps/unix/sysv/linux/x86_64/clone3.S", line 81, in __clone3 [0x7f695749e84f]
[ros2_control_node-4] #17   Source "./nptl/pthread_create.c", line 442, in start_thread [0x7f695740cac2]
[ros2_control_node-4] #16   Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f695769d252, in 
[ros2_control_node-4] #15   Object "/opt/ros/humble/lib/controller_manager/ros2_control_node", at 0x58a7e3ae108d, in 
[ros2_control_node-4] #14   Object "/opt/ros/humble/lib/libcontroller_manager.so", at 0x7f6957ada4a4, in controller_manager::ControllerManager::read(rclcpp::Time const&, rclcpp::Duration const&)
[ros2_control_node-4] #13   Object "/opt/ros/humble/lib/libhardware_interface.so", at 0x7f6957276c7c, in hardware_interface::ResourceManager::read(rclcpp::Time const&, rclcpp::Duration const&)
[ros2_control_node-4] #12   Object "/opt/ros/humble/lib/libhardware_interface.so", at 0x7f69572944cf, in hardware_interface::System::read(rclcpp::Time const&, rclcpp::Duration const&)
[ros2_control_node-4] #11   Object "/home/v25/franka_ros2_ws/build/franka_hardware/libfranka_hardware.so", at 0x7f69502127e1, in franka_hardware::FrankaHardwareInterface::read(rclcpp::Time const&, rclcpp::Duration const&)
[ros2_control_node-4] #10   Object "/home/v25/franka_ros2_ws/build/franka_hardware/libfranka_hardware.so", at 0x7f695023f478, in franka_hardware::Robot::readOnce()
[ros2_control_node-4] #9    Object "/home/v25/franka_ros2_ws/build/franka_hardware/libfranka_hardware.so", at 0x7f695023e89f, in franka_hardware::Robot::readOnceActiveControl()
[ros2_control_node-4] #8    Object "/home/v25/franka_ros2_ws/install/libfranka/lib/libfranka.so.0.15.0", at 0x7f6950108de7, in franka::ActiveControl::readOnce()
[ros2_control_node-4] #7    Object "/home/v25/franka_ros2_ws/install/libfranka/lib/libfranka.so.0.15.0", at 0x7f69501068a2, in franka::Robot::Impl::throwOnMotionError(franka::RobotState const&, unsigned int) [clone .cold]
[ros2_control_node-4] #6    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f695766f4d7, in __cxa_throw
[ros2_control_node-4] #5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f695766f276, in std::terminate()
[ros2_control_node-4] #4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f695766f20b, in 
[ros2_control_node-4] #3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f6957663b9d, in 
[ros2_control_node-4] #2    Source "./stdlib/abort.c", line 79, in abort [0x7f69573a07f2]
[ros2_control_node-4] #1    Source "../sysdeps/posix/raise.c", line 26, in raise [0x7f69573ba475]
[ros2_control_node-4] #0  | Source "./nptl/pthread_kill.c", line 89, in __pthread_kill_internal
[ros2_control_node-4]     | Source "./nptl/pthread_kill.c", line 78, in __pthread_kill_implementation
[ros2_control_node-4]       Source "./nptl/pthread_kill.c", line 44, in __pthread_kill [0x7f695740e9fc]
[ros2_control_node-4] Aborted (Signal sent by tkill() 4729 1005)
[ERROR] [ros2_control_node-4]: process has died [pid 4729, exit code -6, cmd '/opt/ros/humble/lib/controller_manager/ros2_control_node --ros-args -r __ns:=/ --params-file /tmp/launch_params_kf8cahe5 --params-file /home/v25/franka_ros2_ws/install/franka_fr3_moveit_config/share/franka_fr3_moveit_config/config/fr3_ros_controllers.yaml -r joint_states:=franka/joint_states'].
[INFO] [launch]: process[ros2_control_node-4] was required: shutting down launched system
[INFO] [franka_gripper_node-7]: sending signal 'SIGINT' to process[franka_gripper_node-7]
[INFO] [joint_state_publisher-5]: sending signal 'SIGINT' to process[joint_state_publisher-5]
[INFO] [move_group-3]: sending signal 'SIGINT' to process[move_group-3]
[INFO] [robot_state_publisher-2]: sending signal 'SIGINT' to process[robot_state_publisher-2]
[INFO] [rviz2-1]: sending signal 'SIGINT' to process[rviz2-1]
[franka_gripper_node-7] [INFO] [1750927566.591001348] [rclcpp]: signal_handler(signum=2)
[move_group-3] [INFO] [1750927566.593316148] [rclcpp]: signal_handler(signum=2)
[robot_state_publisher-2] [INFO] [1750927566.594684899] [rclcpp]: signal_handler(signum=2)
[rviz2-1] [INFO] [1750927566.596313171] [rclcpp]: signal_handler(signum=2)
[move_group-3] [INFO] [1750927566.608791295] [moveit.ros_planning_interface.moveit_cpp]: Deleting MoveItCpp
[move_group-3] [INFO] [1750927566.609364485] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Stopped publishing maintained planning scene.
[move_group-3] [INFO] [1750927566.609844325] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Stopping world geometry monitor
[... cut ...] file /tmp/launch_params_4hcp6bit --params-file /tmp/launch_params_9315cw56'].
[move_group-3] 
[ERROR] [rviz2-1]: process has died [pid 4723, exit code -11, cmd '/opt/ros/humble/lib/rviz2/rviz2 -d /home/v25/franka_ros2_ws/install/franka_fr3_moveit_config/share/franka_fr3_moveit_config/rviz/moveit.rviz --ros-args -r __node:=rviz2 --params-file /tmp/launch_params_wrnks6s1 --params-file /tmp/launch_params_bncc1595 --params-file /tmp/launch_params_xejco0ei --params-file /tmp/launch_params_3lwtcem4'].

leobarcellona, 2025-07-02:
Hello, we are encountering the same problem: after a some minutes moveit crashes (libfranka: Move command aborted: motion aborted by reflex! ["communication_constraints_violation). Only with moveit. For example, if we launch franka_bringup we are not getting the error. 
We already changed the network card and the robot is directly connected to the network card. We have also changed the cpu profile to performance. 

The franka research 3 system version is 5.8. The libfranka version 0.15.0. 

@VaRuN-2509 have you found a solution?

SKaveendya27, 2026-03-13:
@VaRuN-2509 @leobarcellona have you found a solution for this?