frankarobotics/franka_ros2 #165: gripper

joonsu0109gh, 2025-07-16:
[image]

[image]

My FR3 gripper is properly initialized, but when I run
ros2 launch franka_fr3_moveit_config moveit.launch.py robot_ip:=172.16.0.2,
the state of the end-effector shows "Not connected". Although I can operate the fr3_arm, the gripper planning appears to work in RViz but does not affect the real robot.
Could you help me resolve this issue?

ignacio-pm, 2025-08-01:
Hi,
in order to identify the problem causing the gripper not working, I suggest sharing the results of the following tests:

1. Use the gripper launch file without moveit to move the gripper: `ros2 launch franka_gripper gripper.launch.py`
2. Move the gripper through Desk through one of these options:
- If you have the App Package installed, use the Move or Pick app
- If you do not have the App Package installed, you can move the gripper using the pilots buttons: First, press the Pilot-Mode button (button on the top of the arrows, see section 6.1 of the [FR3 product manual](https://download.franka.de/documents/Product%20Manual%20Franka%20Research%203_R02210_1.0_EN.pdf)) and then press the arrow keys (see section 11 of the [Franka Hand manual](https://download.franka.de/documents/220010_Product%20Manual_Franka%20Hand_1.2_EN.pdf).

If none of these options work, please share your system version and if you have experienced this bug with other system versions.

ignacio-franka, 2026-09-30:
Hi @Nat-J ,
do you still have this issue?
As I proposed in my previous comment, you can test the command without moveIt `ros2 launch franka_gripper gripper.launch.py` to isolate the problem. If it is a bug in the moveit_config, I will forward it so that we can take a look.

AndreasKuhner, 2026-11-12:
Closing this issue:

1. **MoveIt gripper controller name mismatch (`Action client not connected to action server: fr3_gripper/gripper_action`):** Fixed in `v3.5.0`. `franka_fr3_moveit_config/config/fr3_controllers.yaml` and `moveit.launch.py` now match the `franka_gripper` node name launched by `gripper.launch.py`.
2. **End-effector showing "Not connected" in Desk / `UDP receive: Timeout`:** If the gripper still shows as disconnected in Desk when FCI is active, please follow the hardware/system troubleshooting steps above or reach out to `support@franka.de` with the robot logs.

closed 2026-11-12 by AndreasKuhner